awabom/3dprint #3: Full install not changing anything

phoenX80, 2023-09-17:
I come from Cura, that wasn't able to create satisfactible prints anymore and instantly ran into the issues you are fixing with your scripts. Thanks for your help to resolve this! I followed your "all scripts" guide, installing Python 3.11.5 and copying the scripts. But I didn't notice any change in the print behaviour. I selected the SV04 once more and nothing changed. Then I copyied all your script lines from the seperate fixes into the post processing scripts. And sliced again and this time I saw a CMD window several times with no errors sclicing. I will try a new print in 2h approx. But I am afraid, Prusa may be like double applying your scripts this way? How can I tell that just copying the files does do anything? I didn't see a new "SV04" - ist here supposed to be one or how may I tell Prusa is using your mod?

Windows 11 pro, Reooted after Python install and after copying files.

awabom, 2023-09-19:
After copying the prusa profiles, there should be a new printer to select called "SV04 Dual and Single Mode (0.4mm nozzle)" - does this printer appear?

After selecting this printer, you need to select e.g. 'Normal 0.20mm with Scripts' to trigger the scripts.

Do not change the folder structure of the scripts. They need to be in separate folders when you extract the whole repository.

awabom, 2023-10-06:
No reply... closing.

closed 2023-10-06 by awabom

phoenX80, 2023-10-30:
Probably resolved: I did everything correct. But I assumed you ment it will pop up in the config wizard which it doesn't. You meant probably inside the Prusa GUI the "Printer" setting now shows "User presets" and there I can see it. Just a mix up. Thank you for your help and work.

awabom, 2023-11-04:
No problem :)