awabom/3dprint #5: Z-Offset lost

phoenX80, 2023-10-30:
As I finally used the scripts correctly, the printer obviously behaves differently, like wiping twice and prior z axis correction but it seems to loose Z offset completely?  Do I need to adjust the offset in the script or Prusaslicer seperatly?
[](https://youtu.be/nUCudzwLpss[](url))

awabom, 2023-10-31:
Wow... have never had that happen.

I notice that your working g-code does not call M420 S1 after G28 to load the bed mesh? Have you run the mesh leveling and **_saved_** the mesh? I'm not sure if G28 re-enables bed leveling or not in the johncarlson21 firmware.

I must admit, I pretty much only use the Beta-printer profile... Does that one freak out in the same way?

I would check the bed level mesh first though.

phoenX80, 2023-10-31:
Hey, I bed level that thing between every second print and save it. Some times more often. 
As I said, this is imho not about the filament profiles where you offer the beta one but about the selected printer. There ist no beta option of a printer as far as I have seen. I only have time to look for the printer during weekends or less - just so you know I need some time to try things. I will try one of your printers with beta next time.

awabom, 2023-10-31:
Under "Printer Settings" there should be a profile called:

"SV04 Dual and Single Mode - Beta (0.4mm nozzle)"

...and I checked the source code. The firmware does not enable bed leveling on G28 unless it was previously enabled.

So the standard working g-code probably doesn't use bed leveling mesh if I'm correct...

phoenX80, 2023-10-31:
Ok, thanks. I'll try that next time. Probably Friday. Regards.

closed 2023-11-14 by awabom

phoenX80, 2024-06-22:
I didn't print anything since a few months. I had to reinstall my laptop and reinstalled prusa and your scripts.
I have the exact same issue again. Old gcode on SD runs fine, if I select the newly created one, the printer levels like a hundred times and then happily starts to print 1cm in air....omfg. I forgot how I "fixed" the issue. My kid is pressuring me to get his Pikachu started ... any ideas? :)

awabom, 2024-06-22:
You need to compare the start g-codes in the thing that works and the thing that does. So everything dealing with G28, G34, and so on.

phoenX80, 2024-06-22:
If someone could some time tell me what is wrong, to codes look too different for me to try to insert random code:

`; generated by PrusaSlicer 2.7.4+win64 on 2024-06-22 at 07:43:41 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; support material extrusion width = 0.40mm

M73 P0 R235

; printing object PikachuThick_tail.stl id:0 copy 0
; stop printing object PikachuThick_tail.stl id:0 copy 0

;TYPE:Custom
M605 S1 ; Auto-Park Mode
G90 ; use absolute coordinates
M83 ; extruder relative mode
M413 S0 ; disable power-loss recovery
M211 S0 ; disable software end-stop (otherwise left extruder parks at edge of bed when only right extruder is used)

M140 S70 ; set final bed temp

M104 S185 T0 ; set temporary nozzle temp to prevent oozing during homing (or disable if not used)
M104 S215 T1 ; set temporary nozzle temp to prevent oozing during homing (or disable if not used)

M280 P0 S160 ; Reset BL/CR Touch

G28 ; home all axis
G28 X ; home extruder on x-axis (to the buckets)

M190 S70 ; wait for bed temp to stabilize

G34 ; z stepper auto-align
G28 ; home all axis
G28 X ; home extruder on x-axis (to the buckets)
M420 S1 ; enable bed leveling

M104 S215 T0; set final nozzle temp

M104 S245 T1; set final nozzle temp

G1 Z0.3 F7200 ; Move closer to bed

T0

G92 E0 ; Reset Extruder
M211 S0 ; disable software end-stop for T0 to do wipe
G1 X-50 F7200 ; Move to bucket in case it moves to bed
M109 S215 T0; wait for nozzle temp to stabilize
G1 E20 F1200 ; Dump in bucket
G1 X-5 F7200 ; Wipe
G1 X-50 F7200 ; Back to bucket

; Lower temp of nozzle until PrusaSlicer ooze-prevention heats either of the extruders (only if both are used)
; Removed cool-down: M104 S200 T0 ;cooldown

T1

G92 E0 ; Reset Extruder
G1 X350 F7200 ; Move to bucket in case it moves to bed
M109 S245 T1; wait for nozzle temp to stabilize
G1 E20 F1200 ; Dump in bucket
G1 X305 F7200 ; Wipe
G1 X350 F7200 ; Back to bucket

G1 E-5 F1200 ; CUSTOM RETRACT - UNDO T1: G1 E5 F1200 ; OozeFix

; Lower temp of nozzle until PrusaSlicer ooze-prevention heats either of the extruders (only if both are used)
M104 S220 T1 ;cooldown

G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
; tool change

M211 S0 ; disable software end-stop for T0

T0 ; active next extruder
G1 X-50 F7200 ; move to bucket
M109 S208 T0 ; make sure heated (over bucket)

M211 S1 ; enable software end-stop after T0 change

; Filament gcode
M900 K0.05 ; Calibrated using 0.4mm nozzle and HobbyKing PLA (Leaf Green)
M109 S215 T0 ; set temperature and wait for it to be reached
M107
;LAYER_CHANGE
;Z:0.3
;HEIGHT:0.3
; printing object PikachuThick_tail.stl id:0 copy 0
; stop printing object PikachuThick_tail.stl id:0 copy 0
G1 E-1.25 F1200
G1 Z.3 F7200
G1 X112.233 Y115.095
G1 Z.3
G1 E2.25 F1200
;TYPE:Skirt/Brim
;WIDTH:0.45
G1 F1800
G1 X112.981 Y114.383 E.04967
G1 X114.155 Y113.438 E.07249
G1 X115.414 Y112.603 E.07266
G1 X116.501 Y112.002 E.05974
G1 X119.026 Y110.942 E.13171
G1 X123.045 Y109.681 E.20259
G1 X124.391 Y109.354 E.06662
G1 X125.503 Y109.158 E.05431
G1 X127.388 Y108.994 E.091
G1 X166.566 Y109.052 E1.88433
G1 X168.618 Y109.201 E.09895
G1 X170.743 Y109.601 E.104`

VS old code

`;FLAVOR:Marlin
;TIME:7649
;Filament used: 2.57539m, 0m
;Layer height: 0.1
;MINX:112.065
;MINY:110.832
;MINZ:0.3
;MAXX:188.817
;MAXY:184.433
;MAXZ:36.5
;Generated with Cura_SteamEngine 5.4.0
T0
M82 ;absolute extrusion mode
;SV04 start
M140 S60;
M104 S200;
M280 P0 S160;
G4 P100;
G28;
M420 S1;
M190 S60;
M109 S200;
G92 E0;
G1 X10.1 Y20 Z0.28 F5000.0;
G1 X10.1 Y200.0 Z0.28 F1500.0 E15;
G1 X10.4 Y200.0 Z0.28 F5000.0;
G1 X10.4 Y20 Z0.28 F1500.0 E30;
G92 E0 ;Reset Extruder
G1 Z2.0 F3000;
G92 E0
G92 E0
G1 F2400 E-0.5
;LAYER_COUNT:363
;LAYER:0
M107
M204 S500
M205 X16 Y16
G0 F2880 X125.328 Y120.942 Z0.3
M205 X8 Y8
;TYPE:SKIRT
G1 F2400 E0
G1 F1200 X127.874 Y118.855 E0.24636
G1 X129.891 Y117.106 E0.44615
G1 X131.903 Y115.26 E0.65049
G1 X133.116 Y114.078 E0.77724
G1 X134.05 Y113.192 E0.87358
G1 X134.616 Y112.705 E0.92946
G1 X135.387 Y112.17 E0.99968
G1 X136.183 Y111.732 E1.06768`

phoenX80, 2024-06-22:
Hey,
I once again didn't really understand what you meant, but now I did and it works.
The issue was to choose the "dual and single mode BETA" PRINTER in PrusaSlicer.
The machine atm prints normally, I wonder what to do for COPY mode. I will have to not use your scripts there? Not that often using it.